maglub/piLogger #16: getTemperature does not fetch temperatures for aliases.

do3meli, 2015-01-12:


maglub, 2015-01-19:
Could you please send me output of:

```
~/piLogger/bin/dbTool -a
~/piLogger/bin/getTemperature YourAliasName
```

do3meli, 2015-01-19:
```
pi@raspberrypi ~ $ ~/piLogger/bin/dbTool -a
device ID       | device alias
----------------+---------------
28.777043050000 | TempInside
28.5F8D43050000 | TempOutside
pi@raspberrypi ~ $ ~/piLogger/bin/getTemperature TempInside
24.13
pi@raspberrypi ~ $ ~/piLogger/bin/getTemperature TempOutside
2.00
```
seems to work on the develop branch:
```
pi@raspberrypi ~/piLogger $ git status
# On branch develop
```

maglub, 2015-01-19:
Can you reproduce the bug in master?

maglub, 2015-01-19:
I merged develop with master. So this should be fixed somehow now.

closed 2015-01-19 by maglub