maglub/piLogger #22: unable to open db file during db setup.sh

do3meli, 2015-01-14:
while running setup.sh the first time i saw these errors:

```
Setting up sqlite3 (3.7.13-1+deb7u1) ...
  - Setting upp /var/piLogger/db/app.sqlite3 database
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
/home/doemeli/piLogger/bin/functions-sqlite3: line 416: [: -eq: unary operator expected
    - Table device does not exist -> creating
      - Creating table device from ddl.sql
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
/home/doemeli/piLogger/bin/functions-sqlite3: line 416: [: -eq: unary operator expected
    - Table alias does not exist -> creating
      - Creating table alias from ddl.sql
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
/home/doemeli/piLogger/bin/functions-sqlite3: line 416: [: -eq: unary operator expected
    - Table devicegroup does not exist -> creating
      - Creating table devicegroup from ddl.sql
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
/home/doemeli/piLogger/bin/functions-sqlite3: line 416: [: -eq: unary operator expected
    - Table plotgroup does not exist -> creating
      - Creating table plotgroup from ddl.sql
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
/home/doemeli/piLogger/bin/functions-sqlite3: line 416: [: -eq: unary operator expected
    - Table plotconfig does not exist -> creating
      - Creating table plotconfig from ddl.sql
Error: unable to open database "/var/piLogger/db/app.sqlite3": unable to open database file
```

maglub, 2015-01-14:
This is a result of the files not being created due to the lack of "sudoers" permissions for your user.

Most likely, we should add a check that it is possible to create the directories in /var in the configure script.

maglub, 2015-01-19:
Added check in "configure" script to warn when the user does not have the proper sudoers configuration.

closed 2015-01-19 by maglub