maglub/piLogger #23: permission to create directory denied during setup.sh

do3meli, 2015-01-14:
during the setup.sh i saw below error
```
mkdir: cannot create directory `/var/piLogger/xml': Permission denied
```

maglub, 2015-01-14:
Your user has to be in the /etc/sudoers file or member of the unix group "sudo".

Perhaps I should add a check in the configure script for sudoers rights.

```
maglub@domopi:~/piLogger$ grep maglub /etc/group
sudo:x:27:pi,maglub
maglub@domopi:~/piLogger$ sudo -l
Matching Defaults entries for maglub on this host:
    env_reset, mail_badpass, secure_path=/usr/local/sbin\:/usr/local/bin\:/usr/sbin\:/usr/bin\:/sbin\:/bin

User maglub may run the following commands on this host:
    (ALL : ALL) NOPASSWD: ALL
```

do3meli, 2015-01-14:
not sure if this is the problem as my user was always able to do sudo. only difference was that i had to enter a password once when i started setup.sh and the first sudo command run:
```
Matching Defaults entries for doemeli on this host:
    env_reset, mail_badpass, secure_path=/usr/local/sbin\:/usr/local/bin\:/usr/sbin\:/usr/bin\:/sbin\:/bin

User doemeli may run the following commands on this host:
    (ALL : ALL) ALL
```
i have changed the `/etc/sudoers` file now and `sudo -l`showing the following:
```
User doemeli may run the following commands on this host:
    (ALL : ALL) ALL
    (ALL) NOPASSWD: ALL
```
i am bit confused now why i have 2 entries here...

maglub, 2015-01-14:
Most likely since you are both a member of the group "sudo", which per
default has (ALL : ALL) ALL, and that you have an entry with your username
in /etc/sudoers that is "(ALL) NOPASSWD: ALL"

do3meli, 2015-01-14:
good point. i would use this command to see if a user is part of the sudo group
 `grep sudo /etc/group | grep $USER`
unfortunately you need sudo to be able to grep the `/etc/sudoers` file. 
so not sure how you could verify the 2nd way you mentioned.

maglub, 2015-01-14:
:-) "sudo -l" shows what you are allowed to do

maglub, 2015-01-15:
```
pi@domopi ~ $ sudo -l | grep "(ALL.*ALL$"
    (ALL : ALL) NOPASSWD: ALL
    (ALL) NOPASSWD: ALL
pi@domopi ~ $ echo $?
0
```

closed 2015-01-19 by maglub