maglub/piLogger #32: disable any html caching options

do3meli, 2015-01-21:
we need to make sure we have all the caching options disabled with HTML and maybe also the webserver config needs to be looked at. the reason why i am openening this issue is that my home setup http://pilogger.dominicschlegel.ch is missing 35 min of data points when opening from my company's network. if i open it on my mobile it seems to be up to date somehow. I think we need to be able to tell proxies not to cache any data somehow.

do3meli, 2015-01-21:
after clearing the browser cache new data is getting showed. we should adjust lighttpd cache controls as descriped here http://www.cyberciti.biz/faq/lighttpd-send-cache-control-maxage-headers-of-staticfiles/ in the `setup.sh`script

maglub, 2015-01-21:
I enabled "mod_expire" and added the following config:

```
$HTTP["url"] =~ "^/cache" {
       expire.url = ( "" => "access 10 seconds" )
}
setenv.add-response-header += (
      "Cache-Control" => "public, must-revalidate, proxy-revalidate"
)
```

This will force the browser and any proxy to revalidate at least the "/cache" after 10 seconds.

closed 2015-01-21 by do3meli

maglub, 2015-01-23:
malu@KMG-Hotspot-3.v.cablecom.net:/Users/malu/kmg/customers/runesson/priv $curl -I http://pilogger.smalometern.com/cache/sensorAllInfo.json
HTTP/1.1 200 OK
Server: nginx/1.1.19
Date: Fri, 23 Jan 2015 15:26:28 GMT
Content-Type: application/json
Content-Length: 260
Connection: keep-alive
Expires: Fri, 23 Jan 2015 15:26:37 GMT
Cache-Control: max-age=10
Accept-Ranges: bytes
ETag: "1509713007"
Last-Modified: Fri, 23 Jan 2015 15:26:14 GMT
Access-Control-Allow-Origin: http://kmggroup.ch

malu@KMG-Hotspot-3.v.cablecom.net:/Users/malu/kmg/customers/runesson/priv $curl -I http://pilogger.smalometern.com
HTTP/1.1 200 OK
Server: nginx/1.1.19
Date: Fri, 23 Jan 2015 15:26:33 GMT
Content-Type: text/html
Connection: keep-alive
X-Powered-By: PHP/5.4.35-0+deb7u2
Access-Control-Allow-Origin: http://kmggroup.ch