maglub/piLogger #33: listDevices not working anymore

do3meli, 2015-01-21:
the gauges are not getting printed anymore since last release. seems to be an issue with listDevices. further looking it has to do something with the change you did in 2ba642a5d14184a32ae5b73412af689f2daf76ba. it looks like $dataDir variable is empty when listDevices gets called....
```
pi@raspberrypi ~/piLogger/bin $  /home/pi/piLogger/bin/listDevices --json --info 
[ERROR: opening '/db/28.5F8D43050000.rrd': No such file or directory
 
{ "sensorName":"28.5F8D43050000", "devicePath":"/mnt/1wire/28.5F8D43050000" , "aliases":["TempOutside"], "temperature": }ERROR: opening '/db/28.777043050000.rrd': No such file or directory
,
{ "sensorName":"28.777043050000", "devicePath":"/mnt/1wire/28.777043050000" , "aliases":["TempInside"], "temperature": }
]
```

do3meli, 2015-01-21:
actually the data logging isn't working at all anymore

maglub, 2015-01-21:
Ops, bad patch, then.

Will fix it after dinner.

maglub, 2015-01-21:
You are on the right track, but you should remove dataDir from the functions file.

The bug is that I renamed a couple of "global" variables in ~/piLogger/etc/piLogger.conf, which are reflected in ~/piLogger/configure.

So if you remove ~/piLogger/etc/piLogger.conf (or move it), then run:

```
cd ~/piLogger
./configure --withAbioWire --withLocale
```

This will recreate piLogger.conf and the system will start working again.

The idea is that any variable that is global, should be defined in ~/piLogger/etc/piLogger.conf

do3meli, 2015-01-21:
got it - it is working again

closed 2015-01-21 by do3meli