maglub/piLogger #37: multiple sensors @ 1 device

do3meli, 2015-01-29:
If you have 1 device that offers the ability to read multiple sensor data we need to come up with better design as it is currently only possible to read 1 sensor data from a device even if it has more than 1. For example the humidity sensor from Axiris http://www.axiris.eu/en/index.php/one-wire/one-wire-humidity-sensor offers the ability to read temperature and humidity via owfs:

```
pi@raspberrypi ~/piLogger/bin $ cat /mnt/1wire/26.A1E97B000000/temperature 
     24.6562
pi@raspberrypi ~/piLogger/bin $ cat /mnt/1wire/26.A1E97B000000/humidity 
     22.3825
```

my idea is to have another table in the sqlite3 database that looks something like this:
```
create table sensors (id text, sensortype text);
```
in my example the id would be = `26.A1E97B000000` and the sensortype = `temperature` or `humidity`
so finally i would end up having 2 rows for that device:

This would then mean that we should create the .rrd files based on the data available in the sensors table instead of the device table. Also we could use the sensors table to join it with the devices table on column id. that way we can find out if a device has multiple sensors attached. 

i did not yet finish thinking around this design but i think that may be a good start.

maglub, 2015-02-03:
I agree having more configuration options in the database is ok, not sure exactly what yet.

For the rrd database files, I would suggest that we just add some sort of identifier to the rrd files, so that device 26.A1E97B000000 having both a humidity and temperature sensor would get two rrd-files:

26.A1E97B000000.temperature.rrd
26.A1E97B000000.humidity.rrd

This way we don't have to recreate files if/when we figure that we would need even more columns/sensor types. It will require a few changes in the function files, and that we rename existing database files to include {temperature, humidity}.

maglub, 2015-05-08:
First step, renaming the old rrd-files and adding "default" type "temperature" to scripts.

Next step is to:

* Create a generic "logMetric" script instead of the "logTemperature" script
* Fix REST API to catch the metric type as well as the metric and device name

maglub, 2015-10-01:
In some of the later commits, I have added functionality to:

* Automatically add new rrd-database files for different metric types
* Automatically log the proper metric type in the right rrd-database

What is now missing is:
* the REST api entry to log correctly
* rewrite the scripts ./logAll and ./logTemperature to use the correct metric type