maglub/piLogger #50: install owfs from package list source

do3meli, 2015-05-08:
currently the owfs package gets manually downloaded and installed by the setup.sh script. i found out that the owfs packages exist in the official raspberry package list so it could be installed with apt.

* remove current owfs download in setup.sh
* add apt-get install in AbioWire section of setup.sh
* create cleanup-upgrade patch to remove the AbioWire directory as it is no longer needed

do3meli, 2015-05-08:
see the found owfs packages:
```
pi@domopi2 ~/piLogger/AbioWire $ sudo apt-cache search owfs
libow-perl - Dallas 1-wire support: Perl5 bindings
libow-php5 - Dallas 1-wire support: PHP5 bindings
libow-tcl - Dallas 1-wire support: Tcl bindings
owfs - Dallas 1-wire support
owfs-common - common files used by any of the OWFS programs
owfs-dbg - Debugging symbols for the OWFS packages
owfs-doc - Dallas 1-wire support: Documentation for owfs
owfs-fuse - 1-Wire filesystem
python-ow - Dallas 1-wire support: Python bindings
ufsutils - UFS filesystems utilities
```

do3meli, 2015-05-08:
attention this will may install unwanted owhttpd or owftpd packages:
```
pi@domopi3 ~ $ sudo apt-get install owfs
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libcryptsetup4 libgssglue1 libsystemd-daemon0 libsystemd-id128-0 libsystemd-journal0
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  fuse-utils libow-2.8-15 owfs-common owfs-fuse owftpd owhttpd owserver
Suggested packages:
  owfs-doc
The following NEW packages will be installed:
  fuse-utils libow-2.8-15 owfs owfs-common owfs-fuse owftpd owhttpd owserver
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 394 kB of archives.
After this operation, 1,163 kB of additional disk space will be used.
Do you want to continue [Y/n]? n
```

maglub, 2015-05-08:
It might be that OWFS support AbioWire nowadays. When I created the
setup.sh script the bundled owfs did not.

We can easily either ignore that the owhttpd and owftpd or disable them
after install, if we go for the normal repo owfs, which is of course the
preferred way to go if the AbioWire card works with it.

do3meli, 2015-05-08:
i did not check for the AbioWire support...

do3meli, 2015-05-09:
AbioWire uses 1-Wire I2C host controllers (2 x DS2482-800 + 1 x DS2482-100) as per:
http://www.axiris.eu/en/index.php/1-wire/abiowire 

i have checked owfs official page and they seem to support these 2 different controllers:
http://owfs.org/index.php?page=i2c-ds2482-800
http://owfs.org/index.php?page=i2c-ds2482-100

unfortunately i do not have a spare AbioWire Board and i dont want to test it with my existing setup :smile:

closed 2015-05-09 by do3meli