maglub/piLogger #62: Log file permissions

maglub, 2015-06-16:
the log files in /var/log/piLogger are owned pi:pi with chmod 644

This means that when the web server run bin/logTemperature, the process is owned by  it does not log to the log file, although it saves the data in the rrd and can use the remote-logging.d scripts

do3meli, 2015-08-04:
@maglub are you suggesting to change the chmod to something else like 646?

maglub, 2015-08-04:
I would say that if others have rw, it doesn't matter what you have for group permissions.

I would rather make sure the logfiles have a group ownership that both the webserver and pi can write to, alternatively (also not optimal) run the web server as pi.

Or have the web server run the scripts using sudo -u pi.

maglub, 2015-11-24:
This is obsolete now, since the web gui does not run external scripts anymore.

closed 2015-11-24 by maglub