maglub/piLogger #75: setup.sh should check for raspi-blacklist.conf file

do3meli, 2015-08-27:
while running the setup script on a brand new raspberry pi i got the following error:

```
  - i2c config in /etc/modprobe.d/raspi-blacklist.conf
sed: can't read /etc/modprobe.d/raspi-blacklist.conf: No such file or directory
  - adding i2c-dev to /etc/modules
  - setting up locale (if not already set up)

``
why is this file not existing on my installation? maybe there should be a check if that file exists or not before doing a sed...

do3meli, 2015-08-27:
it seems raspian devs have removed this file some time ago as quickly mentioned in this stackoverflow question here: http://stackoverflow.com/questions/21028227/raspberry-pi-does-not-have-etc-modprobe-d-raspi-blacklist-conf

do3meli, 2015-08-27:
it seems that as the blacklist file got removed the `setup.sh` script is not activating the I2C kernel driver. the recommendation is to do this via `sudo raspi-config` but i dont know how this could be programmed...

do3meli, 2015-08-27:
seems like they moved the configuration for loading kernel at startup to `/boot/config.txt` so we could adjust this file instead.

maglub, 2015-08-28:
Good call. Quite a lot has happened in the raspbian distro since i got started.

maglub, 2015-08-31:
http://wiki.m.nu/index.php/OWFS_på_Rasperry_Pi

(in swedish)

commit b2762b41 (2015-09-06): setup now using /boot/config.txt instead of raspi-blacklist.conf - fixes #75
closed 2015-09-06 by do3meli