maglub/piLogger #78: Why is setup.sh using dpkg -s for installation of new packages?

do3meli, 2015-08-27:
why is the setup.sh using `dpkg -s` for installation of new packages and not directly `sudo apt-get install`. i dont see the point of having an additional lookup if a package is installed before actually doing it. apt itself checks if a package is already installed or not as you can see from below output:

```
pi@domopi3 ~/piLogger $ sudo apt-get install php5
Reading package lists... Done
Building dependency tree       
Reading state information... Done
php5 is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

all lines that are starting with dpkg like the one below:
```
sudo dpkg -s lighttpd >/dev/null 2>&1 || { echo "  - Installing lighttpd" ; sudo apt-get -y install lighttpd ; }
```
could be re-written to:
```
echo "  - Installing lighttpd"
sudo apt-get -y -qq install lighttpd
```
this way we could improve the script by not using any pipe or redirect.

maglub, 2015-08-28:
It is for pure cosmetic reasons. The dpkg -s alternative is silent; no "installing..." message unless the package really is installed.

Not completely sure if there also is a time gain, since the dpkg -s will tell you immediatelly if the package is installed. The "||" is a logical "or", not a pipe.

do3meli, 2015-08-28:
Using the -qq argument in apt-get is doing a silent installation too😉no message at all except if errors occuring.

dpkg -s is faster if the package is already installed...

maglub, 2015-08-28:
In this example the "  - Installing lighttpd" is always echoed, even if
nothing is installed:

```
echo "  - Installing lighttpd"
sudo apt-get -y -qq install lighttpd
```

In the current script the echo is done only if the package is
installed. I vote for keeping the "dpkg -s", especially if it is also
faster if the package is already installed.

do3meli, 2015-08-28:
Ok. I agree. Keep the faster option with dpkg. Still one improvement could be to check all packages that are getting installed with the apt-get install. For example: the current php check is greping for php5. That could be improvet to include every single php5 package we install, like php5-cgi, php5-cli, ect...😉

maglub, 2015-10-01:
Updated the setup.sh script with a for loop for the php packages, checking each and every one of them before installing.

closed 2015-10-01 by maglub