maglub/piLogger #100: ddl.sql will break fresh installation

do3meli, 2015-09-20:
if one does install a fresh new pi and starts the setup.sh with the current version of the `ddl.sql` the upgrades that run afterwards will break the system most likely.

one idea would be to pre-insert a certain version to the version table. but we do have to make sure all the changes up on the latest release are included in the `setup.sh` script.

maglub, 2015-09-20:
Hmm, the issue is even trickier.

We need to figure out when running setup.sh if it is really run for the first time. If this is the case, we can inject the version number in "currentVersion" into the database.

If there is no /var/lib/piLogger/db/app.sqlite3 file, it is created in setup.sh, which could also be an indication that the script is run for the first time.

maglub, 2015-09-20:
Resolved through 1a9866170687c102be3df952af1f3b26b097606e

closed 2015-09-20 by maglub