maglub/piLogger #98: upgrade should check for fails

do3meli, 2015-09-20:
we do upgrade the version even if an upgrade fails. we may should not do that and exit the upgrade immediately not that the upgrade after the failed one will be executed.

maglub, 2015-09-20:
Actually, we do not upgrade the version if the upgrade fails.

In the piLogger_upgrade() function in functions-upgrade, we check the rc:

```
  piLogger_doUpgrade_${fromVersion}_to_${toVersion}
  local rc=$?

  [ $rc -eq 0 ] && { piLogger_setInstalledVersion "$toVersion" ; echo "    - Success" ; }
  [ $rc -ne 0 ] && { echo "    - FAILED" ; return 1 ; }
```

We should take care and return the proper return code from the upgrade functions.

do3meli, 2015-09-20:
should we then maybe update the existing upgrade functions and make sure they return status codes correctly?

maglub, 2015-11-24:
I would say that since we have gotten much better at error handling in later releases, we should just close this ticket and aim to take care of exceptions in the functions-upgrade and upgrade.sh from now on.

closed 2015-11-24 by maglub