maglub/piLogger #104: use different javascript sources based on environment

do3meli, 2015-09-24:
use CDN javascript ressources if it is a production environement.
use local javascript ressources if it is a local environement.
that could be achieved with the stub file

maglub, 2015-09-24:
Given that we would have a variable set (i.e $localCDN = true;), how would we go at this in the Twig template?

do3meli, 2015-09-24:
we could define 3 different variables/arrays in `index.php`:

```
$twig->addGlobal('useCDN', true);
$twig->addGlobal('localJS',array( 'path/to/JS/nr/1', 'path/to/JS/nr/2' ));
$twig->addGlobal('CDNJS',array( 'path/to/CDN/JS/nr/1', 'path/to/CDN/JS/nr/2' ));
```
then we could remove all javascript definitions in the `_layout.html` file and add the following:

```
{% if useCDN  %}
   {% for res in CDNJS %}
        <script res="{{ res }}" type="javascript"></script>
    {% endfor %}
{% else %}
      {% for res in localJS %}
        <script res="{{ res }}" type="javascript"></script>
    {% endfor %}
{% endif %}
```

maglub, 2015-10-23:
Opted for a solution using piLogger/etc/piLogger.conf as control mechanism, and a variable called "isOffline".

```
#-------------------------
# Development status
#-------------------------
isOffline="true"
```

To get this into php, I added a function which reads in this config file:

```
$config = getAppConfig($root . "/../etc/piLogger.conf");
...
$twig->addGlobal('isOffline', (isset($config['isOffline']) && $config['isOffline'] == "true")?true:false);
```

commit 3e72f08a (2015-10-23): Fixes #104. Added configuration option to use local resources vs CDN shipped js/css, for when developing in a vm when not connected to a network
closed 2015-10-23 by maglub