maglub/piLogger #106: show amount of undelivered piLogger-cloud data

do3meli, 2015-09-24:
show somewhere on the web interface how much files are still in the spool directory

maglub, 2015-09-24:
Counting >100k files in the spool directory takes time, and could become a DDOS possibility if counted every time a webpage is loaded.

Perhaps is done by cron.

```
pi@1wire-r001fry /var/log/piLogger $ time find /var/spool/piLogger/ -type f 2>/dev/null| wc -l
113245

real	0m1.857s
user	0m0.700s
sys	0m1.190s
```

do3meli, 2015-09-24:
can you check timings for something like this in PHP for example:

```
echo count(array_diff( scandir('/var/spool/piLogger/'), array('.', '..') ));
```

maglub, 2015-09-24:
```
pi@1wire-r001fry ~/test $ cat test.php 
<?php
echo count(array_diff( scandir('/var/spool/piLogger/'), array('.', '..') ));
?>
pi@1wire-r001fry ~/test $ time php test.php 
109849
real	0m2.537s
user	0m2.140s
sys	0m0.390s
```

do3meli, 2015-09-24:
even worse :-1:

maglub, 2015-09-24:
```
pi@1wire-r001fry ~/test $ time ls -f /var/spool/piLogger/ | wc -l
109034

real	0m0.784s
user	0m0.450s
sys	0m0.340s
```

maglub, 2015-09-24:
```
pi@1wire-r001fry ~/test $ cat dircnt.c 
#include <stdio.h>
#include <dirent.h>

int main(int argc, char *argv[]) {
    DIR *dir;
    struct dirent *ent;
    long count = 0;

    dir = opendir(argv[1]);

    while((ent = readdir(dir)))
            ++count;

    closedir(dir);

    printf("%s contains %ld files\n", argv[1], count);

    return 0;
}

pi@1wire-r001fry ~/test $ gcc -o dircnt dircnt.c 
pi@1wire-r001fry ~/test $ time ./dircnt /var/spool/piLogger/
/var/spool/piLogger/ contains 108098 files

real	0m0.351s
user	0m0.080s
sys	0m0.250s
```

do3meli, 2015-09-24:
another language for this project? i think the ls option is "good" enough?

maglub, 2015-09-24:
Nah, I just wanted to see what the fastest way to count files was. Just for reference.

Found that code snippet online somewhere.

do3meli, 2015-09-24:
so you are happy with the `ls` option if i would implement that?

maglub, 2015-09-24:
sure, why not. if it breaks at some point we can always remove it

commit 131072eb (2015-10-30): added info column for plugin config - close #106
closed 2015-10-30 by do3meli