maglub/piLogger #111: Test piLogger on raspian jessie

do3meli, 2015-09-30:
Raspian today released a new version of their OS based on debian jessie. We have to make sure it runs on jessie as smooth as on wheezy. A how-to -upgrade link is mentioned in the following article: http://www.golem.de/news/raspberry-pi-neue-raspbian-version-jessie-1509-116590.html

maglub, 2015-09-30:
Quote: Die Macher hinter Raspbian empfehlen eine komplette Neuinstallation des Betriebssystems statt eines Updates. Eine nichtoffzielle Anleitung für ein Update findet sich jedoch im Raspberry-Pi-Forum.

* Upgrade path "Neuinstallation": run ./piLogger/bin/backup.sh, copy the /var/lib/piLogger/backup/<date>_<time>.tgz, install new SD card, install piLogger from scratch (including setup.sh), copy tgz file to the raspberry pi, "cd / ; tar xvzf /path/to/file.tgz", ./piLogger/bin/refreshCaches

* Upgrade path "Upgrade per inofficial description": Has to be tested

do3meli, 2015-09-30:
Will the newly installed OS have the same device ID?

maglub, 2015-09-30:
Should have, since the device ID is calculated from the MAC address.

I have a device here in the office I can test it on.

//magnus

maglub, 2015-10-01:
My upgrade of d1wire-r001fry went well, and it seems like I have no issues. Should we make a note in README.md about the upgrade path, and close this ticket?

do3meli, 2015-10-01:
my fresh new installation yesterday went well as well. i think it is good to close this issue. i think we dont have to mention the upgrade path in `README.md`. every user should decide on their own how to do this. in addition the software should not be bound to a specific OS or any version of a operating system i think ;-)

closed 2015-10-01 by do3meli