maglub/piLogger #120: Copy ssh-keys in backup.sh

maglub, 2015-10-01:
When making a backup, copy the content of .ssh as well.

do3meli, 2015-10-01:
my security ring bell is going on like crazy ;-) i think this is maybe not a such a good idea as one would not expect to have a backup file containing ssh private keys.

maglub, 2015-10-01:
Hmmm, you've got a very good point. But what about your Mac's time-machine backups? Don't they contain your ssh-keys?

I.e, these backup files are owned by the user "pi", and if you get to these files, you already compromized the system and have access to ~/.ssh

I can live without these backed up. It is just annoying, since I forgot to bring over the ssh-keys from my development raspberry pi, and have to go through the 10 seconds to add new keys in github. =)

maglub, 2015-10-01:
Agreeing with you, skipping this enhancement.

closed 2015-10-01 by maglub