maglub/piLogger #121: Setup.sh does not set the right permissions for /var/lib/piLogger/db

maglub, 2015-10-02:
The directory /var/lib/piLogger/db should have the group ownership of "www-data" and the permission 775.

```
pi@1wire-r002stg ~ $ ls -la /var/lib/piLogger/db
total 162068
drwxrwxr-x 2 pi www-data     4096 Sep 29 08:35 .
drwxr-xr-x 8 pi pi           4096 Jun 16 18:09 ..
-rw-rw-r-- 1 pi www-data 10370124 Oct  2 09:40 28.20F3AD030000.temperature.rrd
```

This is not set up that way in setup.sh

maglub, 2015-10-02:
Quick workaround:

```
sudo chgrp www-data /var/lib/piLogger/db
sudo chmod 775 /var/lib/piLogger/db
sudo chgrp www-data /var/lib/piLogger/db/*
sudo chmod g+w /var/lib/piLogger/db/*
```

do3meli, 2015-10-25:
do you really mean the group ownership or the owner? in the above example the .rrd file is already having a group ownership of www-data. on my installation i am also having group ownership of www-data for all .rrd files:

```
pi@domopi2 ~ $ ls -la /var/lib/piLogger/db/*.rrd
-rw-rw-r-- 1 pi www-data 10370124 May  8 10:06 /var/lib/piLogger/db/26.A1E97B000000.humidity.rrd
-rw-rw-r-- 1 pi www-data 10370124 Oct 25 14:01 /var/lib/piLogger/db/26.A1E97B000000.temperature.rrd
-rw-rw-r-- 1 pi www-data 10370124 Oct 25 14:01 /var/lib/piLogger/db/28.5F8D43050000.temperature.rrd
-rw-rw-r-- 1 pi www-data 10370124 Oct 25 14:01 /var/lib/piLogger/db/28.777043050000.temperature.rrd
```

so in my point of view there is no further change needed. www-data should be able to read and write to the files in its current chmod of 664.

maglub, 2015-10-26:
This is the problem on a new install:

https://github.com/maglub/piLogger/blob/master/setup.sh#L119

```
[ ! -d "$dbDir" ]      && { echo "Creating dbDir: $dbDir"           ; sudo mkdir -p "$dbDir"      ; }
...
sudo chown ${myUser}:${myGroup} "$dbDir"
```

commit 2e7c6331 (2015-10-26): Changed ownership and permissions on the db directory, so that the web server can access it. Resolves #121
closed 2015-10-26 by maglub