maglub/piLogger #128: add drag & drop functionality

do3meli, 2015-10-12:
add drag & drop functionality so one could move a chart from one place to another.

maglub, 2015-11-29:
Commit fb70baba74b5a4c5e0899da3be157749a37ed43e deals with this in https://github.com/maglub/piLogger/blob/master/include/templates/index.html and https://github.com/maglub/piLogger/blob/master/include/templates/macros.html

* I had to simplify the div structure in the macro.html, so that the graph box now has one div-level less.
* Dragging a graph, dropping it on another graph, will put the source graph before the destination graph.

Even if this is now pushed into master, there is a bit of cleanup to do:

"class: column" should be renamed to something more descriptive, and corresponding names in the javascript code should be updated
"class: arne-dase" must for sure be renamed to something better

maglub, 2015-11-29:
To further make this persistent in the database, we will need to:

1) Give the div id's unique names, so that we can reference exactly which graph was put where in which order
2) Add a rest call to reflect this, and update the database

At the moment, we also have more than one group of graphs, where the concept of what to show when is not very thought through. The idea was to be able to define multiple dashboards, and have multiple groups of plots. At the moment, we display all plots in the table "plotconfig" on the main page index.html, whereas we probably should only display the "default" plots. Other names in the "name" column of the plotconfig table should only be displayed when explicitly requested.