maglub/piLogger #138: resetPassword does not accept any characters

do3meli, 2015-10-26:
the resetPassword function does not accept any characters. for example the resetPassword script will return an error if for example the password contains a closing bracket ")" or something like this "::H!".

maglub, 2015-10-26:
Works fine, you just have to use ""

```
 ./resetPassword "::H!"
```

Otherwise the shell will expand the character strings.

do3meli, 2015-10-26:
ok - we will have to mention this somewhere in the readme ;-)

maglub, 2015-10-26:
Should be part of standard unix knowledge, but ok; never hurts to be clear.

Example:

```
maglub@ubuntu-14:~/dev/www/piLogger/bin$ echo ::H!)
-bash: !: event not found
maglub@ubuntu-14:~/dev/www/piLogger/bin$ echo "::H!"
::H!
maglub@ubuntu-14:~/dev/www/piLogger/bin$ echo "::H!)"
-bash: !: event not found
maglub@ubuntu-14:~/dev/www/piLogger/bin$ echo '::H!)'
::H!)
```

commit 8a5e7d1e (2015-10-26): More precise description on how to set new password, resolves #138
closed 2015-10-26 by do3meli