maglub/piLogger #139: Bug: setup.sh installs apache2 together with php5

maglub, 2015-10-29:
When setting up a new system, the setup.sh installs apache2 together with one of the php packages.

This breaks the installation after a reboot, since apache2 is started before lighttpd.

maglub, 2015-10-29:
Quote:

You can get around this by installing one of the other dependencies before php5. For example:

```
apt-get install php5-fpm php5 

or 

apt-get install php5-cgi php5.
```

do3meli, 2015-10-29:
could we not disable the auto startup for apache2?

maglub, 2015-10-29:
Sure we could, but would only be a workaround and a waste of both bandwidth and space.

We can avoid installing it by using the construct above, installing php5-cgi before php5, see the referenced link.

maglub, 2015-10-29:
Installing php5-cgi before php5 solved this.

maglub, 2015-11-15:
New install still install apache2.

commit 97ce354f (2015-11-24): Resolves #139. Disabling apache2
commit 4a5b91a4 (2015-11-24): Resolves #139
closed 2015-11-24 by maglub