maglub/piLogger #143: Disable 1-wire TCP server, http daemon, ftp server in setup.sh

maglub, 2015-11-12:
No need for anything but /mnt/1wire. The other features should be disabled

do3meli, 2015-11-13:
how could that be achieved the best way? 
removing the files in `/etc/init.d/` directory or adjusting the file `/etc/owfs.conf `

maglub, 2015-11-13:
sudo update-rc.d owftpd disable
sudo update-rc.d owhttpd disable
sudo update-rc.d owserver disable

do3meli, 2015-11-13:
on my setup here these ow* services are not even running so i am wondering if that is really needed:

```
pi@domopi3 ~/piLogger $ netstat -an | grep 2121 | grep -v grep
pi@domopi3 ~/piLogger $ netstat -an | grep 2120 | grep -v grep
pi@domopi3 ~/piLogger $ netstat -an | grep 4304 | grep -v grep
```
the only thing running is owfs:
```
pi@domopi3 ~/piLogger $ ps -ef | grep ow | grep -v grep
root       605     1  0 Oct21 ?        00:20:25 /usr/lib/owfs/owfs -c /etc/owfs.conf --i2c=ALL:ALL --allow_other /mnt/1wire/
```
is that different on your setup?

maglub, 2015-11-13:
I mainly see them when I reboot a rpi and have a scren/console connected.

They are not needed, sp we could just as well make sure to disable them.

commit 09c4230f (2015-11-13): disabling unused owfs components - close #143
commit d61a6d4b (2015-11-13): Merge pull request #145 from do3meli/disable-owfs-components

disabling unused owfs components - close #143
closed 2015-11-13 by do3meli