maglub/piLogger #144: Bug: piLogger-cloud process spool dir fails when input file is 0 bytes

maglub, 2015-11-12:
The script dies with the cryptic message:

```
20151112_214849;piLogger-cloud;Error: could not connect to api.pilogger.net to send file /var/spool/piLogger/piLogger-cloud_30457_1447344324
```

When the 0 byte input file is removed, the remaining files are sent to pilogger.net

do3meli, 2015-11-13:
so the question is: why are there files with 0 byte? 
further i am wondering if they get generated with 0 bytes or if they get sent and the deletion is not working properly. i could have imagined that if 2 process try to send it that there is a situation where a 0 byte file could be left over. but we always check if spoolDir process runs or not before calling it...

maglub, 2015-11-14:
The question is valid (why are there zero bytes files), but the case still has to be dealt with. There can be many different reasons, not mutually exclusive, to why there are zero bytes files.

maglub, 2015-11-14:
For example, this would be an ok way to deal with it:

```
maglub@ubuntu-14:~/dev/www/temp/piLogger$ git diff remote-logging.d/piLogger-cloud
diff --git a/remote-logging.d/piLogger-cloud b/remote-logging.d/piLogger-cloud
index e7edb33..b204573 100755
--- a/remote-logging.d/piLogger-cloud
+++ b/remote-logging.d/piLogger-cloud
@@ -114,6 +114,8 @@ function processSpoolDir(){
      sensor=$( cat $file | cut -d"," -f 1)
      time=$( cat $file | cut -d"," -f 2)
      value=$( cat $file | cut -d"," -f 3) 
+        
+        [ -z "$sensor" ] && { logIt "  - Error: $file does not contain any data, removing it." ; rm $file ; continue ; }
 
      response=$(curl -H 'Content-Type: application/json' -X PUT \
                 --write-out "%{http_code}" --silent --output /dev/null \
```

do3meli, 2015-11-14:
Perfect. I see you you have it already. So commit it and good is😉Thanks

commit c514f180 (2015-11-14): remove empty spool files - close #144
closed 2015-11-14 by do3meli