maglub/piLogger #146: setup does not set correct permission for /var/spool/piLogger

do3meli, 2015-11-23:
on a fresh installation the /var/spool/piLogger directory is having the following permissions:
```
drwxr-xr-x 2 root root 4096 Sep 30 16:20 .
```
with these permissions i cant log any files...

maglub, 2015-11-24:
On a fresh installation, this should be done properly. https://github.com/maglub/piLogger/blob/master/setup.sh#L118

```
sudo chown ${myUser}:${myGroup} "$logDir"
```

maglub, 2015-11-24:
I think you must have run setup.sh like "sudo ./setup.sh" to get that effect.

Perhaps we should change the assignment of $myUser (https://github.com/maglub/piLogger/blob/master/setup.sh#L114,L115)

Perhaps we should force it to be "pi:pi", like:

```
myUser=pi
myGroup=pi
```

do3meli, 2015-11-24:
Hmm...i never re-run setup.sh on this paricular pi. So we did not properly cover it with upgrades. And it doesnt make sense now to create a new version that is somewhere in the middle of others...

maglub, 2015-11-24:
Ok, that could probably be it. So, this might already be solved, then.

But we should have something in cron that is checking the configuration, so that we can warn for such things, and also notify about new releases.

do3meli, 2015-11-24:
Before we close this would you be able to do a fresh installation an check our assumption? I never run setup with sudo...

maglub, 2015-11-24:
New install:

```
pi@dashboard-r001mue ~/piLogger $ cat currentVersion 
20151113-001
pi@dashboard-r001mue ~/piLogger $ ls -la /var/log | grep piLogger
drwxr-xr-x  2 pi       pi         4096 Nov 24 12:07 piLogger
```

closed 2015-11-24 by maglub