maglub/piLogger #147: piLogger-cloud spool data for sensors that are not registered with pilogger.net

maglub, 2015-11-24:
piLogger-cloud cannot send sensor data to pilogger.net for devices that are not registered. This can happen when:

* One add more sensors to the device after the registration process, or
* One use ./logMetric to add new sensor data for "virtual" or non existing sensors ad hoc for non registered sensors

```
pi@d1wire-r001fry ~/piLogger/bin $ ./logMetric arne 123
20151124_143047;logMetric;  - Logging data for device arne, 123 C
20151124_143047;logMetric;    - callRemoteLoggingPlugins: found piLogger-cloud
20151124_143047;piLogger-cloud;error: did get HTTP status code 403 from REST api - something went wrong
20151124_143047;piLogger-cloud;writing measured data to /var/spool/piLogger/piLogger-cloud_2608_1448371847 for later processing
20151124_143047;logMetric;    - callRemoteLoggingPlugins: found piLogger-remote
20151124_143047;piLogger-remote;Logging arne:123 to 192.168.2.61
{"sensorId":"arne","temperature":"123","ok":true,"msg":"Added temperature to the database. Return message: OK!"}200
20151124_143047;logMetric;    - callRemoteLoggingPlugins: found shiftr.io
20151124_143048;shiftr.io;  - Sending data to shiftr.io: d1wire-r001fry:arne:123
OK
pi@d1wire-r001fry ~/piLogger/bin $ ls -latr /var/spool/piLogger/
total 296
drwxr-xr-x 6 root root   4096 Oct  1 18:38 ..
-rw-r--r-- 1 pi   pi       62 Nov 24 14:30 piLogger-cloud_2608_1448371847
drwxrwxrwx 2 pi   pi   290816 Nov 24 14:30 .
pi@d1wire-r001fry ~/piLogger/bin $ cat /var/spool/piLogger/piLogger-cloud_2608_1448371847 
a39c1872c47ab8fad315f90a4ac5cd9c,2015-11-24 14:30:47,123,arne
```

* Log entry from piLogger-cloud for this error:

```
20151124_143047;piLogger-cloud;error: did get HTTP status code 403 from REST api - something went wrong
```

We should:

* Add better logging for piLogger-cloud
* Not spool data for non registered sensors (or make it an opitonal configuration)

maglub, 2015-11-24:
When running the curl command manually, without output to /dev/null, we get a proper error message from pilogger.net, so that side is taking care of it properly.

```
 curl -H 'Content-Type: application/json' -X PUT                 --write-out "%{http_code}" --silent                 -d "{\"probeTime\": \"2015-11-24 14:43:23\",\"probeValue\": \"123\",\"authToken\": \"XYZ\"}"                 http://api.pilogger.net/api/sensor/XYZ

{"error":true,"msg":"this sensor is not allowed to store any new data points with the given authToken"}403
```

do3meli, 2015-11-24:
We could compare the given sensor in piLogger-cloud to what is actually attached to the device. If it is attached then spool the files otherwise do not save the files or do not even send it to the cloud - independent of the response. What do you think?

maglub, 2015-11-24:
I am almost done testing another solution.

The easier solution is to actually trust pilogger.net, which sends a http 403 back, when a sensor is not allowed to be sent to pilogger.net.

Using this (403), I can just:

1) Not spool the metric
2) If already spooled, delete it

do3meli, 2015-11-24:
Nice idea. I like it. But it will send unused data over to amazon which may will cause additional unneeded costs😉

maglub, 2015-11-24:
We can extend it further on to cover that use case as well (which is in the same direction as in blacklisting devices).

maglub, 2015-11-24:
see 181f4f1

do3meli, 2015-11-24:
should we keep this issue open for the second part with blacklisting devices or should we create a new one for that?

closed 2015-11-24 by do3meli