maglub/piLogger #150: remote logging should have a blacklist concept

do3meli, 2015-11-24:
the remote logging concept should have a blacklist concept that allows to exclude a certain alias or device id from being passed on to the enabled plugins. the benefit would be that no unneeded data is being sent over to other services.

maglub, 2015-11-24:
The blacklist concept should probably be on a "per plugin" basis, since the
only plugin that has an issue so far, is the pilogger.net

do3meli, 2015-11-27:
i challenge that ;-) 
in my view it could be also interesting for the remote logging plugin. a possible use case may be that a sensor is broken and sending wrong data. with a blacklist that data is kept locally (maybe for analysis of the problem useful) but never sent to plugins (which assume the data delivered is correct)

maglub, 2015-11-27:
So, my suggestion is a file $baseDir/etc/piLogger.yml

This file should have the possibility to blacklist sensors either globally,
or per plugin. I have been playing with the thought to convert the config
file into YML, to simplify the cross-language usage (bash/php), so the
blacklist could be a first test to see if it would work.

Example $baseDir/etc/piLogger.yml

```
pi@d1wire-r001fry ~/piLogger/etc $ cat piLogger.yml
interface:          AbioWire

piLoggerLocale:     true

baseDir:            /home/pi/piLogger
dataDir:            /var/lib/piLogger
oneWireDir:         /mnt/1wire

htmlDir:            $baseDir/html
cacheDir:           $dataDir/cache
temperatureFile:    temperature
rrdDir:             $dataDir/db
dbDir:              $dataDir/db
graphDir:           $dataDir/graphs
backupDir:          $dataDir/backup
spoolDir:           /var/spool/piLogger

appDbFile:          $dbDir/app.sqlite3

logDir:             /var/log/piLogger

#----------------------
dateFormat:         "+%Y%m%d %H%M%S"

#-------------------------
# Parallel processing
#-------------------------
parallelProcessing: 4

#-------------------------
# for remote logging
#-------------------------
#--- remoteLogging=true
remoteLogging="true"
remoteHost="192.168.2.61"

remoteLogging:
  enabled:       true
  host:          192.168.2.61

  blacklist:
    global: 28.D14443050000
    piLogger_cloud: 28.D14443050000

#-------------------------
# for piLogger Cloud
#-------------------------
  piLogger_cloud:
    deviceIdentifier: SECRET
    authToken: "SECRET"

#------------------------
# shiftr.io
#------------------------
  shiftr:
    shiftrAuth: "956eccaee9827fc6:851145836ca5c152f649f67de914dd01"
```

And the parser, parse_yml, to use for bash scripts:

```
#!/bin/bash

# From:
http://stackoverflow.com/questions/5014632/how-can-i-parse-a-yaml-file-from-a-linux-shell-script

this_dir=$(cd `dirname $0`; pwd )
this_script=$(basename $0)

function usage(){
  cat<<EOT

Usage:
  $this_script filename.yml

EOT
}

function parse_yaml {
#   local prefix=$2 #--- not used at the moment.
   local printSection=$2
   local prefix=""

   local s='[[:space:]]*' w='[a-zA-Z0-9_]*' fs=$(echo @|tr @ '\034')
   sed -ne "s|^\($s\):|\1|" \
        -e "s|^\($s\)\($w\)$s:$s[\"']\(.*\)[\"']$s\$|\1$fs\2$fs\3|p" \
        -e "s|^\($s\)\($w\)$s:$s\(.*\)$s\$|\1$fs\2$fs\3|p"  $1 |
   awk -F$fs '{
      indent = length($1)/2;
      vname[indent] = $2;
      for (i in vname) {if (i > indent) {delete vname[i]}}
      if (length($3) > 0) {
         vn=""; for (i=0; i<indent; i++) {vn=(vn)(vname[i])("_")}
         printf("%s%s%s=\"%s\"\n", "'$prefix'",vn, $2, $3);
      }
   }'
}

[ -z "$1" ] && { usage ; exit 0 ; }

parse_yaml $1 $2 | grep ^$2
```

Output from this script, if we onyl want to show the section "remoteLogging"

```
pi@d1wire-r001fry ~/piLogger/etc $ ../bin/parse_yml ./piLogger.yml
 remoteLogging
remoteLogging_enabled="true"
remoteLogging_host="192.168.2.61"
remoteLogging_blacklist_global="28.D14443050000"
remoteLogging_blacklist_piLogger_cloud="28.D14443050000"
remoteLogging_piLogger_cloud_deviceIdentifier="cbf5f22f726a1b3eecab58fe6050bdec"
remoteLogging_piLogger_cloud_authToken="82cf1e294e0bd84967560a7d2f6e294f"
remoteLogging_shiftr_shiftrAuth="956eccaee9827fc6:851145836ca5c152f649f67de914dd01"
```

In a bash script, i.e the piLogger-cloud, it could be used as:

```
$($baseDir/bin/parse_yml $configDir/piLogger.yml remoteLogging)
```

Would this be a way? That way, we can use the two variables
$remoteLogging_blacklist_global and $remoteLogging_blacklist_piLogger_cloud
to decide what to not send upstream. Sadly, one cannot use "-" in a
variable name, but with some quick parsing, we could make the last part of
the config the script name.

Best,
//magnus

maglub, 2015-11-27:
Eh, the markdown on the comment above is correct, but it does not display correctly. Weird.

Well, you get the basic idea.

maglub, 2015-11-27:
Another example:

```
pi@d1wire-r001fry ~/piLogger/etc $ ../bin/parse_yml ./piLogger.yml remoteLogging_blacklist
remoteLogging_blacklist_global="28.D14443050000"
remoteLogging_blacklist_piLogger_cloud="28.D14443050000"
```

maglub, 2015-11-27:
Testing new feature in 3232665

closed 2015-12-11 by do3meli