maglub/piLogger #168: Bug: Raspbian allocate both fixed and DHCP IP

maglub, 2015-12-03:
In the case, when a user would like to set a fixed IP (i.e in /etc/network/interfaces), Raspbian still ask for, and get a DHCP address. I found this out first by looking at the device table of my router, and found that one of my Raspberry Pis had two IP addresses.

It is not visible by only using "ifconfig", but both IP addresses works when connecting to the system.

```
pi@d1wire-r001fry /etc/network $ cat /etc/network/interfaces
auto lo

iface lo inet loopback

allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf

auto eth0
iface eth0 inet static
	address 192.168.2.62
	netmask 255.255.255.0
	network 192.168.2.0
	broadcast 192.168.2.255
	gateway 192.168.2.1
	# dns-* options are implemented by the resolvconf package, if installed
	dns-nameservers 192.168.2.1

#iface default inet dhcp
```

* ifconfig

```
pi@d1wire-r001fry /etc/network $ ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:d8:be:48  
          inet addr:192.168.2.62  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fed8:be48/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:60820 errors:0 dropped:17957 overruns:0 frame:0
          TX packets:17402 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5543821 (5.2 MiB)  TX bytes:2072026 (1.9 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:8 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:480 (480.0 B)  TX bytes:480 (480.0 B)
```

* ip addr show eth0

```
pi@d1wire-r001fry /etc/network $ ip addr show eth0
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:d8:be:48 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.62/24 brd 192.168.2.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet 192.168.2.201/24 brd 192.168.2.255 scope global secondary eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:fed8:be48/64 scope link 
       valid_lft forever preferred_lft forever
```

do3meli, 2016-06-15:
Have you tried this before or after a reboot? I expierienced very funny ip config behaviours on debian prior a reboot and after a interface change. Particular with ipv6 stuff. I could imaging the same may happens with ipv4...

maglub, 2016-06-15:
Yes, I have. And it is due to a not so easy to get around quirk with Raspbian.

I wrote a summary on the previous version of Raspbian, which is similar but that has a couple of other quirks as well, here: http://www.kmggroup.ch/?p=638

It is due to how "dhcpcd" is configured per default. An interface can be ignored, or one can disable dhcpcp.